TomWhitwell/Spring-Reverb-Mk-2 #1: Low Volume / Short When Brick Inserted

junge1987, 2017-11-28:
I have double-checked all values and re-flowed all solder joints as well as verified OP-AMP orientation and checked for solder bridges.  Jumper is inserted in Brick Mode.  I have swapped TL074, TL072, and 78L05 with my private stock to verify functionality of IC's.  I have also verified orientation of Electrolytics. 

When Belton Brick is inserted, it shorts out my rack's power supply upon power up (cycles on and off).

When I remove the Belton Brick, it power up (LED On, Signal Flowing).  Output is extremely low (like less than 5% of original)

Thonk-Steve, 2017-11-30:
Hi, do you also have a real spring tank?

junge1987, 2017-11-30:
I unfortunately do not.  Both of my guitar amps don't have built-in tanks either.

TomWhitwell, 2017-11-30:
This sounds obvious, but do you have the Brick the correct way around - matching the silkscreen on the PCB

junge1987, 2017-11-30:
Yes.  I have it the right way matching the screen.