TomWhitwell/Spring-Reverb-Mk-2 #4: Works great with tank, brick causes low-speed oscillation and distortion

rfcampbell, 2018-01-17:
Not using expander, jumper set correctly to "brick" position.

Wet and dry outputs both involve a fluctuating, heavily distorted sound. 

https://youtu.be/AgYhhPH3KQ8

TomWhitwell, 2018-01-17:
Sorry to hear that, I've never heard one sounding like that before. Go through some standard troubleshooting - check all the solder points, check orientation of chips and the brick, check the voltages with a multimeter. My hunch is that it is a soldering issue somewhere. You may be able isolate the trouble point by checking the x-fade is working as expected - that rules out a certain portion of the board.

rfcampbell, 2018-01-18:
Thanks. Although the joints all looked good, I touched them up and now it works as it should. Sorry for the trouble!

closed 2018-01-18 by rfcampbell