TomWhitwell/Spring-Reverb-Mk-2 #5: No signal from input/feedback, either dry or to tank

danismithcheapdeals, 2018-02-27:
Hi Thom,

I built the spring reverb last week but can't get any dry/wet signal from the input. I can hear the tank itself if I move it or turn up the trimmer to oscillation. X-fade input seems to work. I've tried swapping over the ICs and checked over the solder joints to see if any were bridged/cold. I've had a look over the schematics to try and see what the signal path is. 

Details of the problem:
-I get regular signal through x-fade, can be turned down using wet control
-I get VERY quiet, nearly inaudible signal through Feedback input
-I get basically inaudible/no signal through input.

-I get signal FROM the tank but nothing going in
-If I turn the trimmer to the point of oscillation and then slightly backwards I can hear Feedback
signal faintly through the spring 

-The dry/wet control only seems to work as a Wet/xfade level - NO dry signal at all
-LED on front panel works, not sure if vactrol is working

-no response from feedback or tilt controls

The fact that the crossfader only works on one side makes me think there is an issue with the vactrol, do you have any ideas about what might be the issue? 

Thanks, 
Dani

oliverlyu, 2018-07-10:
I have the exact same issue, any thoughts as to what this might be?

danismithcheapdeals, 2018-07-10:
Have been trying to get this to work on and off over the last few months but nothing yet. If anyone has any suggestions would be greatly appreciated!