TomWhitwell/Spring-Reverb-Mk-2 #6: Distorted output and no Reverb

danfastlikeflan, 2018-05-06:
Hi Tom,
Hopefully you can provide some insight. I have already reflowed all joints and checked for solder bridges. Here are some sypotoms:

-Module turns on and behaves the same with the jumper in either position and without the jumper on at all 
-The led is max brightness with all knobs full clockwise and then gets very dim when the cv knob is about 10% and then will get progressively brighter when rotating the main knob
- with the feedback switch off, the output is very quiet and get louder until feedback is turned up to about 10% and turns into distortion any farther than that.
- The three feedback switches do provide three different tones
-the wet output is always completely quiet
-tilt has no noticeable effect 

Any help is appreciated. 
Thanks,
Daniel

TomWhitwell, 2018-05-06:
That's quite a baffling selection of symptoms. 
The main LED sounds correct 
You can check the crossfade independent of the spring reverb by patching another signal into x-fade and checking that crossfade is smooth. 
Tilt acts on the send to reverb, so should only be audible in reverb tails 
Distortion - have you calibrated the module - setting the trimmer so that it doesn't distort 

Are you using Brick or Spring? 

Jumper = no impact = that suggests a short or missing connection somewhere - perhaps upload high res pictures of the front and back of the boards.

danfastlikeflan, 2018-05-06:
Thanks for the quick reply. Here's a link to some pics: https://imgur.com/a/YYnqVJq.
I have tried both the spring and the brick but neither resulted in any reverb. Even plucking the tank does produce any sound. I tried testing just the crossfade but its still very distorted there is nothing audible until feedback is turned up a little, even with the feedback switch in the off position. The trim pot does not have any effect on the sound. After playing a little more with it, it seems like inverted feedback and no feedback sound the same.

TomWhitwell, 2018-05-06:
OK, so global problems like that might be... 
An IC that is the wrong way around - perhaps a TL072 on the front (panel) board? 
An IC that is not properly seated - so one leg isn't making a connection 
A power issue; a missing/reversed diode. 
You can see the schematic here: http://musicthing.co.uk/collateral/SpringReverbMk2-Schematic.pdf

danfastlikeflan, 2018-05-06:
One of the diodes was backwards. Thanks so much for the help!

closed 2018-05-07 by danfastlikeflan