TomWhitwell/Spring-Reverb-Mk-2 #12: no reverb

mach-spring, 2020-09-06:
Hi, my module looks fine - LED looks good (fades in as I turn the big knob). But there is no reverb. When I turn the big knob the signal out just becomes weaker (to about half). When Tilt is turned up also the sound slightly changes. Thanks for your help!

TomWhitwell, 2020-09-06:
Hi, sorry to hear that. 
Check you have installed the jumper and that the connections to the spring tank look good.

mach-spring, 2020-09-06:
Hi, thanks for your quick answer. 

Could it be because the voltage regulator got too hot?

I measured it though: with red to ground both sides (input and output) read appr. 0.66.

TomWhitwell, 2020-09-06:
Ok, so you definitely have a power problem - you should have +12 on one side and +5 on the other

TomWhitwell, 2020-09-06:
Ok, so you definitely have a power problem - you should have +12 on one side and +5 on the other. Is the polarity correct - is it definitely the 78L05, not a transistor?

mach-spring, 2020-09-08:
Ok. I rebuilt the back board and this one now works so I’m a bit closer to know where to look. I don’t really know how to measure the ‚special‘ components so I’ll start exchanging things. I’ll close this thread as soon as the culprit is found.