TomWhitwell/Spring-Reverb-Mk-2 #13: Module made a spark and now  working

smnkjr, 2021-02-04:
[image]
[image]
[image]

TomWhitwell, 2021-02-04:
Hi, sorry to hear that - it's always hard to diagnose that kind of issue. 
You could start with the TL074 on the back near the phono jacks. If you don't have a spare, you could swap the two TL074 chips to see what happens - one does the drive/recovery, the other does the crossfade circuit. 
However, that doesn't address the original no reverb problem - what tank are you using? Do you also have the 'brick'?

smnkjr, 2021-02-05:
Hi Tom

Thank you for replying this quickly :)
I don't know the brand of the tank, but it isn't anything special. I don't have the 'brick'. I think it's quite possible that the issue with the lack of reverb, came from the tank / cable, so I would be sure that there was a problem with the module, before i tried exchanging the other parts.

I tried swapping the TL074's with no luck. Would it be common for them to get broken from a shortcurcuit? And in that case, would It make sense to buy some new ones?
I checked with my voltmeter, and it made a beep when i connected the +12v with -12v, but no connection between -12v/+12v and ground. 

Cheers

TomWhitwell, 2021-02-05:
Did you buy the tank separately? There are quite specific requirements for the tank - details here: 
https://medium.com/music-thing-modular-notes/everything-i-know-about-spring-reverb-1fb4b32abf87
and here:
https://musicthing.co.uk/pages/spring.html

The chips could potentially be damaged if the short circuit sent a high voltage somewhere it wasn't expected. 

The +12 - -12 connection doesn't sound good at all, but maybe remove all the the chips and separate the boards, then do the same test. A fried chip might have a short inside it, potentially.

smnkjr, 2021-02-05:
I had the tank laying around already. But now I think I will get a new tank instead, that fits the specifications.

I just made some progress! I tried removing both th TL074's, and that removed the +12 - -12 connection. Then I found out that one of the TL074's made the connection, but the other one didn't. 
I'll order some new TL074's from thonk right away, and try it out with a different tank :)

TomWhitwell, 2021-02-05:
Great, that sounds promising - some spring tanks have quite different ground connections, for example, which makes them not work properly. I'll close this now, but come back if you still have trouble.

closed 2021-02-05 by TomWhitwell